ARMmbed/ble #57: add support for Eddystone

prusnak, 2015-07-22:
Having a service similar to ble/services/iBeacon.h but for Eddystone would be nice.

Eddystone protocol specification is here:

https://github.com/google/eddystone/blob/master/protocol-specification.md

rgrover, 2015-07-22:
we're working on it.

rgrover, 2015-08-07:
@BlackstoneEngineering : could you please update and publish a demo on mbed.org around Eddystone. Please note that EddystoneService has now been added to ble/services/

BlackstoneEngineering, 2015-08-07:
Waiting on @rgrover, I can release the preliminary version, but I would really like to use pull request https://github.com/ARMmbed/ble/pull/61 to make an eddystone service that plays nice with other services. The current example clobbers any other service on the device.

rgrover, 2015-08-08:
There's an initial Eddystone service in BLE_API. It was borrowed from work done by @BlackstoneEngineering. I made some changes to the initialization code (removed ble.shutdown() and ble.init()); it is very likely that the current state needs some more work and testing before it can be published as a demo.

BlackstoneEngineering, 2015-08-10:
I have removed the changes (as they broke the service, will revisit at a later date to re-impliment) and submitted a pull request here: https://github.com/ARMmbed/ble/pull/66

BlackstoneEngineering, 2015-08-12:
Here ya go : https://developer.mbed.org/teams/Bluetooth-Low-Energy/code/BLE_EddystoneBeacon_Service/

rainierwolfcastle, 2015-10-15:
http://jira-dev0.cambridge.arm.com/browse/IOTSFW-861

BlackstoneEngineering, 2015-10-15:
@rainierwolfcastle this is a public facing repository, please dont post Jira tickets here.

rgrover, 2015-10-16:
the eddystone service is now a part of the standard services supported by BLE_API.

closed 2015-10-16 by rgrover

prusnak, 2015-10-16:
Great! Thanks!