ARMmbed/ble #58: add central role for s130

fabiencomte, 2015-07-22:
It should be nice if we can use all roles of s130 at the same time.

rgrover, 2015-07-23:
BLE API already allows devices to become a central and use client-APIs. What exactly to is this issue about?

rgrover, 2015-07-23:
https://developer.mbed.org/forum/team-63-Bluetooth-Low-Energy-community/topic/16517/

closed 2015-07-23 by rgrover

fabiencomte, 2015-07-23:
In this forum, someone said that it was not ready.
https://developer.mbed.org/questions/54431/nRF51-S130-central-role-and-roadmap/
So all is ready now ?

fabiencomte, 2015-07-23:
and from https://developer.mbed.org/teams/Bluetooth-Low-Energy/wiki/mbed-BLE-Roadmap

Switching Bluetooth Versions

Top of our list of priorities for the coming releases is extending our BLE APIs to support Bluetooth 4.1 and 4.2. This is expected to bring several improvements:
Support for multiple concurrent connections to the peripheral, and being able to act as a peripheral and central at the same time.

rgrover, 2015-07-24:
yes, the item "support for multiple concurrent connections to the peripheral, and being able to act as a peripheral and central at the same time" is now available. We will create a demo for this shortly.

fabiencomte, 2015-07-25:
Hello,
Demo will be very useful. 
Thank you for your work.

rgrover, 2015-07-26:
Thanks for appreciating.
We're doing our best to accommodate the need for more demos with other re-architecture work. Should anyone from the community offer any demos, it would be delightful.