ARMmbed/ble #69: Infinite busy

fabiencomte, 2015-08-26:
Hello,

I tried this code on 3 boards to make a triangle to test central and device role at same time with S130 on nRF51.

frequently, i loop in busy (2) and never goes out. Maybe i did a mistake, this is my code. 

#include "mbed.h"
#include "BLE.h"
#include "UARTService.h"
#include "ble/DiscoveredCharacteristic.h"
#include "ble/DiscoveredService.h"
#include "UARTService.h"

#define SOFT_DEVICE_FATHER_HANDLE 3

#define BOARDS_COUNT 3

const Gap::Address_t mac_board_0 = {0xb8, 0xac, 0x4e, 0x8d, 0x8b, 0xeb}; 
const Gap::Address_t mac_board_1 = {0x9c, 0x43, 0x62, 0x30, 0xaf, 0xd2}; 
const Gap::Address_t mac_board_2 = {0x5f, 0x1a, 0x9e, 0x6a, 0x63, 0xdd}; 

// tiny ble board
#define LED_GREEN   p21
#define LED_RED     p22
#define LED_BLUE    p23
#define BUTTON_PIN  p17
#define BATTERY_PIN p1

#define MPU6050_SDA p12
#define MPU6050_SCL p13

#define UART_TX     p9
#define UART_RX     p11
#define UART_CTS    p8
#define UART_RTS    p10

DigitalOut led(LED_RED);
DigitalOut alivenessLED(LED_GREEN);
InterruptIn button(BUTTON_PIN);
AnalogIn battery(BATTERY_PIN);
Serial pc(UART_TX, UART_RX);

bool mac_equals(const Gap::Address_t mac_1, const Gap::Address_t mac_2)
{
    #if 0
        if (mac_1[0] != mac_2[0])
        {
            return false;
        }
        if (mac_1[1] != mac_2[1])
        {
            return false;
        }
        if (mac_1[2] != mac_2[2])
        {
            return false;
        }
        if (mac_1[3] != mac_2[3])
        {
            return false;
        }
        if (mac_1[4] != mac_2[4])
        {
            return false;
        }
        if (mac_1[5] != mac_2[5])
        {
            return false;
        }
    #else
[... 369 lines cut ...]
                int ret = ble.gattServer().write(uartServicePtr->getRXCharacteristicHandle(), temp, length);
                if (ret == BLE_ERROR_NONE) 
                {
                    break;
                }
                else if (ret == BLE_STACK_BUSY) 
                {
                    pc.printf("\r\nbusy (2)\r\n");
                    //break;
                }
                else if (ret == BLE_ERROR_OPERATION_NOT_PERMITTED) 
                {
                    pc.printf("\r\nnot connected (2)\r\n");
                    break;
                }
                else
                {
                    pc.printf("\r\ncode %d (2)\r\n", ret);
                }
            }
        }
        
        ble.waitForEvent(); // save power
    }
}

rainierwolfcastle, 2015-10-15:
http://jira-dev0.cambridge.arm.com/browse/IOTSFW-860

jrobeson, 2015-10-16:
@rainierwolfcastle : that url doesn't seem valid. says server doesn't exist

rgrover, 2015-10-16:
@jrobeson that's an ARM internal URL

jrobeson, 2015-10-16:
then they should prefix those messages saying so :)

jrobeson, 2015-10-16:
@fabiencomte: you should post this program as a gist, it'd be a lot easier to read that way. Also, cut it down to the minimal amount of code to reproduce the problem.

rgrover, 2015-10-16:
@jrobeson you're right. they should be marked as ARM internal

0xc0170, 2015-10-16:
Definitely, I thought the plan was to add there internal word. sorry for confusion

rainierwolfcastle, 2015-10-16:
ARM Internal Ref: IOTSFW-1032

rgrover, 2015-10-20:
@fabiencomte is this still an issue for you?

jrobeson, 2016-02-11:
I guess it can be closed then?

fabiencomte, 2016-02-14:
Hello,

Sorry, i missed previous messages notifications. I moved to Nordic SDK to avoid this issue.

fabiencomte, 2016-02-14:
Yes we can close

closed 2016-02-14 by fabiencomte