ARMmbed/ble #83: Advertising data and scan response are committed into Gap object even if the underlying implementation failled

pan-, 2015-10-26:
To illustrate the issue, I will just take the code from Gap::setAdvertisingPayload

```C++
    ble_error_t setAdvertisingPayload(const GapAdvertisingData &payload) {
        _advPayload = payload;
        return setAdvertisingData();
    }

    ble_error_t setAdvertisingData(void) {
        return setAdvertisingData(_advPayload, _scanResponse);
    }

    // this is provided by Gap implementation
    virtual ble_error_t setAdvertisingData(const GapAdvertisingData &, const GapAdvertisingData &) = 0;
```

The member _advPayload is updated even if the *setAdvertisingData* implementation fail. As a result, next calls to getAdvertisingPayload or updateAdvertisingPayload will use or return an invalid advertisingPayload.

Member functions which can commit invalid data are: 
* accumulateAdvertisingPayload
* updateAdvertisingPayload
* setAdvertisingPayload
* accumulateScanResponse

rainierwolfcastle, 2015-10-26:
ARM Internal Ref: IOTSFW-1086

andresag01, 2016-01-19:
The complete list of Gap members that can commit invalid advertising data follows.

* `accumulateAdvertisingPayload(flags)`
* `accumulateAdvertisingPayload(appearance)`
* `accumulateAdvertisingPayloadTxPower(power)`
* `accumulateAdvertisingPayload(type, data, length)`
* `updateAdvertisingPayload(type, data, length)`
* `setAdvertisingPayload(newPayload)`
* `accumulateScanResponse(type, data, length)`

pan-, 2016-04-22:
Fixed by https://github.com/ARMmbed/ble/pull/171.

closed 2016-04-22 by pan-