ARMmbed/ble #86: Wrong include file in HealthTermometerService.h:20:17

LiyouZhou, 2015-10-26:
/work/iot/ble-examples/BLE_Thermometer/yotta_modules/ble/ble/services/HealthThermometerService.h:20:17: fatal error: BLE.h: No such file or directory
 #include "BLE.h"
                 ^
compilation terminated.

rainierwolfcastle, 2015-10-26:
ARM Internal Ref: IOTSFW-1090

rgrover, 2015-10-26:
@LiyouZhou Could you please check with @andresag01 why you are having difficulty building this example?

andresag01, 2015-10-26:
It seems that the include path of BLE.h within the HealthThermometerService.h file should be replaced by "ble/BLE.h".

rgrover, 2015-10-26:
could you please do this, Andres, and submit a pull request?

andresag01, 2015-10-27:
Submitted pull request to 'develop' branch to solve this issue.

rgrover, 2015-10-27:
this will be committed to master with the next release.

commit bde2e465 (2015-10-27): fix #86: update includes from within services to use paths starting from subfolder 'ble'
closed 2015-11-02 by rgrover