ARMmbed/ble #90: BLE::init() should be non-blocking

rgrover, 2015-10-28:
Currently the demo applications call init() and then expect to operate on the BLE instance rightaway.  This model may not apply to stacks where initialization requires several rounds of blocking message passing between the host and the controller. BLE::init() should be non-blocking and should result in a completionCallback() when initialization is complete. It should also be possible for the application to determine if initailization is complete. Any use of a BLE instance before its initialization completes should return an error.

rgrover, 2015-10-28:
!This may result in a rewrite of all demos! The Nordic stack initializes rightaway, and so existing demos should continue to work against Nordic.

mbed-classic demos would look something like:

```
main() {
    BLE::instance().init();
    while (!BLE::instance().hasInitialized()) {
        /* spin wait */
    }

    /* rest of the initialization ending in the waitForEvent loop */

}
```

whereas mbedOS demos would look like:

```
void bleInitComplete(BLE &ble, ble_error_t error)
{
    WsfTrace("bleInitComplete");

    if (error != BLE_ERROR_NONE) {
        WsfTrace("initailization failed with error: %u", error);
        return;
    }

    if (ble.getInstanceID() == BLE::DEFAULT_INSTANCE) {
        /* use the BLE instance */
    }
}

extern "C" void app_start(int argc, char *argv[])
{
    BLE::Instance().init(bleInitComplete);
}
```

rainierwolfcastle, 2015-10-28:
ARM Internal Ref: IOTSFW-1098

commit 5c7f26e0 (2015-10-30): fix #90: add a ble_error_t paramter to initializationCompletionCallback
closed 2015-11-02 by rgrover