ARMmbed/ble #93: Terrible handling of initLen / minLen and variable length characteristics.

Timmmm, 2015-10-30:
See: https://devzone.nordicsemi.com/question/34390/no-ble_gatt_hvx_notification-under-20-bytes/?answer=55637#post-id-55637

Basically there are two problems:

1. The `GattCharacteristic` `initLen` parameter is randomly named `minLen` too. If you follow the code all the way through it actually does two things - it is the initial length (*not* minimum), and it determines whether the characteristic is variable length or not.

2. The characteristic for nRF51822 code is only variable length if `initLen != maxLen`. This is really stupid because you might want the initial length to equal the maximum length but still have variable length. It also isn't documented anywhere.

I suggest the following solution:

1. Add extra constructors with an extra parameter: `bool variableLength`.
2. Name `initLen` to be `initLen` everywhere consistently (and not `minLen`).
3. Deprecate the old constructors.

rainierwolfcastle, 2015-11-02:
ARM Internal Ref: IOTSFW-1159

rgrover, 2015-11-26:
@Timmmm I agree that `initLen` should not have been renamed as `minLen`. I also agree that `initlen != minLen` should not be a criteria for variable-length characteristics.

andresag01, 2015-11-27:
@Timmmm Apologies on the delay for a fix. I have just submitted two pull requests to address the issue (to [ble](https://github.com/ARMmbed/ble/pull/122) and [ble-nrf51822](https://github.com/ARMmbed/ble-nrf51822/pull/78)). The `GattAttribute.initialLen` was removed and we are now using `len` as the current length of `*valuePtr` and `maxLen` as the maximum value of `*valuePtr`. Therefore, we have removed the accessor `getInitialLen()` and replaced this with `getLength()` where required in ble-nrf51822 module.

Notice that currently all attributes are marked as 'variable'. We will be submitting pull requests to deal with this shortly, but it was decided that we would do this separately because it requires changing the constructors of the `GattCharacteristic` and `GattAttribute`.

Timmmm, 2015-11-27:
Awesome, no need to apologise!

andresag01, 2015-12-02:
@Timmmm I have made the latest changes to allow variable length characteristics by passing a boolean to the constructor. Please see: https://github.com/ARMmbed/ble/commit/d71c3161811f126875aa917100967364a303d274.

closed 2015-12-02 by andresag01

Timmmm, 2015-12-02:
Excellent, thanks!