ARMmbed/ble #115: GattCharacteristic size can be optimized by removing read and write authorization callback

pan-, 2015-11-25:
Each `GattCharacteristic` store two FunctionPointersWithContext, one for the readAuthorization callback and one for the writeAuthorization callback. This waste some space given that:
   - In practice these authorizations are not always used (especially the readAuthorization)
   - The implementation of GattServer has to keep track of all GattCharacteristics registered

One way to get rid of these callbacks in Characteristic is to move these callbacks to the GattServer exactly like its already done for onDataRead, onDataWritten or onDataSent.

With this change, only the callbacks requiring authorization are stored in memory and the developer can optimize the amount of authorization callbacks use by grouping some authorization together instead of instantiating one callback by characteristic (for instance, it can be done service wide).

rgrover, 2015-11-25:
thanks for raising this.

marcuschangarm, 2015-12-09:
I'm not particular fond of the current onDataRead, onDataWritten and onDataSent callbacks because all subscribers get notified whether or not the event is relevant for them.

For example, if I have multiple services running at the same time I'm only interested in the data read/write callback for that particular characteristic. 

On the watch, I only use readAuthorization and writeAuthorization instead of onDataRead and onDataWritten.