ARMmbed/ble #121: Radio Notifications API missing functionality?

andresag01, 2015-11-27:
Recently, the Radio notification API was modified to NOT initiate radio notification when a callback is registered using [onRadioNotification](https://github.com/ARMmbed/ble/blob/master/ble/Gap.h#L981). This conforms to the documentation, which now enforces applications to execute something like the following to get Radio Notifications started:
```
ble.gap().onRadioNotification(someHandlerFunction);
ble.gap().initRadioNotification();
```
Nevertheless, the handler only really gets called when there is radio activity, such as after calling `ble.gap().startAdvertising()`. Wouldn't it be best if the radio notification is initiated inside the `onRadioNotification()` call and just modify the documentation to describe this behaviour? After all, you only register a callback because you actually want to get the callback.

Alternatively, if you want to not change the documentation, wouldnt it make sense to have a `stopRadioNotification()` API that allows you to start and stop the radio notification events whenever you want? i.e. even during the advertising process.

pan-, 2015-11-27:
it make sense to register the callback at the point you start the process but I don't think that it belong to the `onRadioNotification` function. Even if we change the documentation, the name doesn't indicate what the function actually does. Plus, sometime you just want to register callbacks at the start of your application then later, depending on the program flow, activate/deactivate the process which trigger this callback.

Maybe we can replicate the `startScan` api.

```c++
ble.gap().startRadioNotification(someHandleFunction);
```

rgrover, 2015-11-27:
yes, I'm happy with setting up a callback along with startRadioNotification()

andresag01, 2015-11-27:
I am happy with the `startRadioNotification()`, but I also think we should have the `stopRadioNotification()`. I think that currently the only way to stop the callbacks would be to call `ble.shutdown()` or perhaps using `onRadioNotification(NULL)` (havent tested this one though).

pan-, 2015-11-27:
Yes, it would be a great addition Andres, API has to be symmetrical. Start/Stop is easy to understand.