ARMmbed/ble #156: AdvertisementCallbackParams_t should include addressType for the peerAddr

rgrover, 2015-12-21:
without this information, a potential central may not be able to connect to a peripheral due to mismatched address-type.

jcady92, 2016-03-24:
Hi Rohit,

Any update on this issue?  I currently am using a gatt client to connect to both Android and iOS devices in the peripheral mode.  It looks like all iOS devices and newer Android devices use random private resolvable addresses.  But some older Android devices (like the Moto G 2014 running 5.0.2) use public addresses.  Right now it's impossible for me to connect to those devices without knowing that the address I receive is in that format.

Any workaround in the meantime?

rgrover, 2016-03-29:
@jcady: my colleague @pan- should be responding to this shortly.