ARMmbed/ble #177: BLE.h conflict with Nordic SDK ble.h

aracosta1, 2016-02-19:
On case insensitive filesystems (Windows first) the two name collide and so often compilation fails because the compiler choose the first of the two file it find in its search path order.

This create big problem when tring to export from mbed online environment to offline build tools like EmBlock or other.

Is it possible to change the mane to something else like BLE_api.h or similar ?

pan-, 2016-02-19:
Hi, 

We all agree that it is not pretty atm. but normally, name should not collides:
* BLE.h from the ble module is included by the directive: `#include <ble/Ble.h>`
* ble.h from the nrf51-sdk module is included by the directive `#include <ble.h>`

aracosta1, 2016-02-20:
I have this problem exporting to Em:block.
The problem is the include statement in the nrf-51 sdk because it does not specify a path s it includes the first ble.h it found in the search path order the is the one from ble_api. I know the correct place to make changes should be the sdk, I'll try to raise the issue on the NRF51822 repository

0xc0170, 2016-03-01:
This was reported on forum.mbed.com, and there was a reference to gcc4mbed, they had the same problem: http://forums.mbed.com/t/error-mbed-nordic-semi-ble-includes-algorithm-into-c-files/1110

pan-, 2016-04-20:
I think this issue has been solved with: https://github.com/ARMmbed/nrf51-sdk/pull/26 which is included in nrf51-sdk version 2.4.0.
@aracosta1 Could you comment on that ?

closed 2016-11-10 by aracosta1