ARMmbed/ble #181: Add error handling to GattClient procedures

pan-, 2016-03-18:
With the current API their is no way to know if an error occur during the following GattClient procedures: 
* `read`: If a read procedure fail, it should be possible to report this information to the application.
* `write`: If a write procedure fail, it should be possible to report this information to the application.

bluetooth-mdw, 2016-08-24:
Totally agree. Was just looking for a way to return an ATT error response in the Application Error range and stumbled across this issue. So it can't be done at present? This seems like a pretty important enhancement.

pan-, 2016-08-24:
@bluetooth-mdw No, it is not yet possible, it will be in a close future. 
This become more and more mandatory. Same thing for the different type of read and write requests to support.

bluetooth-mdw, 2016-08-24:
That's good to hear :-)