ARMmbed/ble #185: Gap.h function updateConnectionParams not implemented

gaucho1978, 2016-07-14:
in repository Bluetooth Low Energy / BLE_API / Gap.h on line 580, the function updateConnectionParams is not implemented. This is needed, when in master role, to reply to the parameter update request coming from the peripherial, according to this chart: http://infocenter.nordicsemi.com/index.jsp?topic=%2Fcom.nordic.infocenter.s130.api.v2.0.0%2Fgroup___b_l_e___g_a_p___c_e_n_t_r_a_l___c_p_u___m_s_c.html&cp=3_6_2_1_0_2_1_3_2

pan-, 2016-07-14:
@gaucho1978 You are looking at the abstraction and not the implementation. 
This function is virtual and is overridden  by implementation.

The implementation on nrf51 is present (see [here](https://github.com/ARMmbed/ble-nrf51822/blob/master/source/nRF5xGap.cpp#L406))

Could you update your ticket if this answer your question ?

gaucho1978, 2016-07-14:
yes, sorry.
I'm continuing the debugging and if i call the following:
 Gap::ConnectionParams_t *connParams;
 connParams->minConnectionInterval=250;
 connParams->maxConnectionInterval=400;
 connParams->slaveLatency=4;
 connParams->connectionSupervisionTimeout=550;
ble_error_t resultCall= ble.updateConnectionParams(connHandle, connParams);
Serial.print("1.Result=" + String(resultCall) + "\r\n");

I Always receive "3" as a result.
 It means that the method is implemented but the parameters seems to be wrong.
 How to find good parameters? these are what the peripherial is requesting me....

gaucho1978, 2016-07-14:
I also add another issue, sorry if i'm making confusion. I updated also the original question.
Read also the following part of the question  please:
"More big problem, is that the peripherial is sending to the master the "connection parameters update request" but the Mbed stack don't implement a callback to inform the user about this request.
I know it because I sniffed the packets.
As a result, after 1 minute the peripherial disconnects from the master."

pan-, 2016-07-14:
@gaucho1978 
* minConnectionInterval: the minimum interval between two connection events 
* maxConnectionInterval: The maximum interval between two connection events
* slaveLatency: The number of connection events a slave can drop
* connectionSupervisionTimeout If no data has been received by the peer in this interval, then the connection is considered over.

The important part here is that: 
connectionSupervisionTimeout > maxConnectionInterval * slaveLatency

So in your case, you can change the connectionSupervisionTimeout to something over 4 * 400.

rogerclarkmelbourne, 2016-07-15:
The other issue with most of the Softdevice wrapper functions is that they do not return the correct response.

In this case the function returns that there has been a paramater error if the SD didn't return success

However there are a load of other possible response codes that the SD could return 
See https://developer.nordicsemi.com/nRF5_SDK/nRF51_SDK_v8.x.x/doc/8.1.0/s130/html/a01260.html#gaf89b434841998ab384e0612dca9e12f8

And at least some of these could be mapped to mbed equivalents, instead of returning a possible erronous error indicating a paramater error when in fact the SD may have actually returned BUSY or that its out of memory etc

gaucho1978, 2016-07-17:
@pan- I think you are wrong because timeout is in steps of 10ms, while min and max inteval are in steps of 1,25ms.

i modified the code to show precise error, according to https://developer.nordicsemi.com/nRF5_SDK/nRF51_SDK_v8.x.x/doc/8.1.0/s130/html/a01260.html#gaf89b434841998ab384e0612dca9e12f8

It turns to be: BLE_ERROR_INVALID_ADDRESS
Effectively, the handle of the connection is 0.
It is zero in the connection callback and also in the discoveryTermination callback.

So the new question is why the handle is zero in the connectionCallBack.

(then the following question it will go back to the original question)

gaucho1978, 2016-07-17:
I checked the Nordic website and it seems normal to have handle equal to zero when the device is in master role (from zero to seven).
So I continue to do not understand why am i getting the NRF_ERROR_INVALID_ADDR

pan-, 2016-07-17:
Oups, my bad, I though the update connection parameters were in ms. 
In some cases APIs they are while in other they're not.  

@gaucho1978 Could you provide a program which reproduce your issue ?

gaucho1978, 2016-07-18:
this is the complete example, consider that it is structured as it follows: on start an active  scan is started. when the desired peripherial is found, it connects, and registers for notifications. On the first notification (inside function hvxCallBack) I request the connection parameters update.

`
#include <BLE_API.h>

BLE           ble;

Gap::Handle_t connectionHandle1;
bool firstNotification=true;

static uint8_t service1_uuid[]    = {0x71, 0x3D, 0, 0, 0x50, 0x3E, 0x4C, 0x75, 0xBA, 0x94, 0x31, 0x48, 0xF1, 0x8D, 0x94, 0x1E};
static uint8_t service1_chars1[]  = {0x71, 0x3D, 0, 2, 0x50, 0x3E, 0x4C, 0x75, 0xBA, 0x94, 0x31, 0x48, 0xF1, 0x8D, 0x94, 0x1E};
static uint8_t service1_chars2[]  = {0x71, 0x3D, 0, 3, 0x50, 0x3E, 0x4C, 0x75, 0xBA, 0x94, 0x31, 0x48, 0xF1, 0x8D, 0x94, 0x1E};
static uint8_t service1_chars3[]  = {0x71, 0x3D, 0, 4, 0x50, 0x3E, 0x4C, 0x75, 0xBA, 0x94, 0x31, 0x48, 0xF1, 0x8D, 0x94, 0x1E};

UUID service_uuid(0x180D);
UUID chars_uuid1(0x2A37);
UUID chars_uuid2(service1_chars2);
UUID chars_uuid3(service1_chars3);

static uint8_t device_is_hrm = 0;
static uint8_t device_is_simple_peripheral = 0;

uint16_t currentHeartRate=0; //variabile che ospita il valore di battito cardiaco ottenuto dal sensore.

// When found the match characteristic, set 1.
static uint8_t characteristic_is_fond = 0;
// When found the match descriptor, set 1.
static uint8_t descriptor_is_found = 0;
// To save the hrm characteristic and descriptor
static DiscoveredCharacteristic            chars_hrm;
static DiscoveredCharacteristicDescriptor  desc_of_chars_hrm(NULL,GattAttribute::INVALID_HANDLE,GattAttribute::INVALID_HANDLE,UUID::ShortUUIDBytes_t(0));

static void scanCallBack(const Gap::AdvertisementCallbackParams_t *params);
static void discoveredServiceCallBack(const DiscoveredService *service);
static void discoveredCharacteristicCallBack(const DiscoveredCharacteristic *chars);
static void discoveryTerminationCallBack(Gap::Handle_t connectionHandle);
static void discoveredCharsDescriptorCallBack(const CharacteristicDescriptorDiscovery::DiscoveryCallbackParams_t *params);
static void discoveredDescTerminationCallBack(const CharacteristicDescriptorDiscovery::TerminationCallbackParams_t *params) ;

/**
 * @brief  Function to decode advertisement or scan response data
 *
 * @param[in]  type            The data type that you want to get
 * @param[in]  advdata_len     The length of advertisement or scan reponse data
 * @param[in]  *p_advdata      The pointer of advertisement or scan reponse data
 * @param[out] *len            If type exist, this is the length of field data
 * @param[out] *p_field_data   If type exist, this is the pointer of field data
 *
 * @return NRF_SUCCESS or NRF_ERROR_NOT_FOUND
 */
uint32_t ble_advdata_parser(uint8_t type, uint8_t advdata_len, uint8_t *p_advdata, uint8_t *len, uint8_t *p_field_data) {
  uint8_t index=0;
  uint8_t field_length, field_type;

  while(index<advdata_len) {
    field_length = p_advdata[index];
    field_type   = p_advdata[index+1];
    if(field_type == type) {
      memcpy(p_field_data, &p_advdata[index+2], (field_length-1));
      *len = field_length - 1;
      return NRF_SUCCESS;
    }
    index += field_length + 1;
  }
  return NRF_ERROR_NOT_FOUND;
}

void startDiscovery(uint16_t handle) {
  /**
[... 395 lines cut ...]
  // put your setup code here, to run once:
  Serial.begin(9600);
  Serial.println("BLE Central Demo ");

  ble.init();
  ble.onConnection(connectionCallBack);
  ble.onDisconnection(disconnectionCallBack);
  ble.gattClient().onServiceDiscoveryTermination(discoveryTerminationCallBack);
  ble.gattClient().onHVX(hvxCallBack);
  ble.gattClient().onDataWrite(onDataWriteCallBack);
  ble.gattClient().onDataRead(onDataReadCallBack);
  // scan interval : in milliseconds, valid values lie between 2.5ms and 10.24s
  // scan window :in milliseconds, valid values lie between 2.5ms and 10.24s
  // timeout : in seconds, between 0x0001 and 0xFFFF, 0x0000 disables timeout
  // activeScanning : true or false
  ble.setScanParams(1000, 200, 0, true);
  // start scanning
  ble.startScan(scanCallBack);
}

void loop() {
  // put your main code here, to run repeatedly:
  ble.waitForEvent();
}
`

gaucho1978, 2016-07-24:
can someone of you try to use the param update and check the result? it's a common problem?

mkarn3yru96, 2020-06-22:
Where does the connHandle comes from?
How is it passed to a variable?