ARMmbed/ble #193: NRF5: Byte order of characteristics with multi-byte data types change

seppestas, 2017-03-28:
Environment: mbed-os 5.4
Device: Ublox NINA-B112 module (based on nRF52832)
Target: UBLOX_EVK_NINA_b1

---

When using a `GattCharacteristic` with a multi-byte value, e.g `uint16_t` or `uint_32_t`, the values are received in an incorrect byte order by the Bluetooth nRF connect app (and probably other consumers of the GATT service).

E.g following characteristics:
```
ReadWriteGattCharacteristic<uint8_t> uint8Characteristic;
ReadWriteGattCharacteristic<uint16_t> uint16Characteristic;
ReadWriteGattCharacteristic<uint32_t> uint32Characteristic;
ReadWriteArrayGattCharacteristic<uint8_t, 4> arrayCharacteristic;

uint8_t uint8_data = 1;
uint16_t uint16_data = 2;
uint32_t uint32_data = 3;

uint8Characteristic(0x00F1, &uint8_data);
uint16Characteristic(0x00F2, &uint16_data),
uint32Characteristic(0x00F3, &uint32_data),
arrayCharacteristic(0x00F4, (uint8_t *)&uint32_data),
```

Results in the following data being presented by the nRF connect app:
![img_0044](https://cloud.githubusercontent.com/assets/1408377/24412413/ad7d8c3c-13d8-11e7-9914-f16fbd4d341c.PNG
This is probably caused by the Nordic S132 softdevice](https://www.nordicsemi.com/eng/Products/S132-SoftDevice) expecting big endian data to be passed to `sd_ble_gatts_characteristic_add`.

It would be nice if the `ReadWriteGattCharacteristic` function could be used to send and receive multi-byte values more easily.

pan-, 2017-03-28:
@seppestas What is incorrect ? Number are transmitted in little endian as expected. 
`nrf Connect` doesn't know oob that the value of characteristics with UUID 0x00F3 is an uint32_t it makes sense if it display it as a succession of byte in the order it was received.

seppestas, 2017-03-29:
@pan- Something is converting the multi-byte values to big-endian. AFAIK the ARMv7-M architecture is little endian by default and so is the Bluetooth L2CAP layer (and thus GATT AFAIK).

Sending the bytes of a buffer in "network byte order", ie Big endian, makes sense, especially when sending things like text. However, it would be nice if the `GattCharacteristic<T>` could be used to send multi-byte data types like ints and floats so we don't have to [shuffle bytes around like in the HeartRateService](https://github.com/ARMmbed/ble/blob/master/ble/services/HeartRateService.h#L159). In my opinion having to deal with byte orders kind of defeats the purpose of having the `GattCharacteristic<T>` templates.

pan-, 2017-03-29:
@seppestas Numbers in BLE are sent in little endian, it is in the specification. 
The uint32_t `3` is stored in memory as `0x03 0x00 0x00 0x00` in little endian. This is exactly what you received in the nrf application. 

About the hearth rate measurement characteristic, the measurement is correctly stored as a little endian value and the bytes are shuffled around because the characteristic value should contain the flags in the first bytes followed by different values configured by the flag ([source](https://www.bluetooth.com/specifications/gatt/viewer?attributeXmlFile=org.bluetooth.characteristic.heart_rate_measurement.xml)).

At the end, I'm not sure to understand what is the issue. Numbers are received by the client in little endian format as expected.

seppestas, 2017-03-29:
@pan- You are right, there is no byte shuffling in the Heart Rate service.I though the endianness of the `hrmCounter` got switched on line 161 and 162 but this indeed does not seem to be the case.

Looks like I was just confused by the way nRF connect displays "byte arrays". I think using a notation like `0x03 0x00 0x00 0x00` is way clearer. Sending 0x00000003 and seeing 0x03000000 appear in the app is a bit strange though.

When "[organizing bytes by byte address](https://en.wikipedia.org/wiki/Endianness#When_organized_by_byte_addresses)", something you (or at least me and ) typically do when dealing with multi-byte data types, you could consider `0x03 0x00 0x00 0x00` to be Big endian... It of course all depends on interpretation.

closed 2017-03-29 by seppestas