ARMmbed/ble #195: X-Nucleo-IDB05A1 Not Working

waymond91, 2017-09-11:
Hello All,
I would like to preface this post by saying that I have absolutely no experience working with the BLE communication stack.
I recently purchased a Nucleo-F446RE board alongside X-Nucleo-IDB05A1 to test out the viability of using the SPBTLE-RF module in a product we are developing. I figured this would be a good way to explore bluetooth within the Mbed environment.
However, when I imported and uploaded the [BLE_HeartRate-IDB0XA1](https://developer.mbed.org/teams/ST/code/BLE_HeartRate_IDB0XA1/) program, as shown in the link, I get absolutely no ouput from the board or the module.
I have tried bluetooth scanning with Android, iOS, Windows, and Linux Devices.
After digging around online a bit, I discovered some people were having success by changing some of the [reset time](https://developer.mbed.org/forum/team-79-ST-community/topic/27616/). I tried this to no available.
A little more probing reveals that there was a firmware update for the ble module that was supposed to make its internal communication stack play nicely with the mbed infrastructure. So I followed the directions [here](https://developer.mbed.org/teams/ST/code/BlueNRG-MS-Stack-Updater/). Although, I only saw the LED blink quickly, indicating an error.
Now my fear is that I've bricked the module by perhaps only performing a partial firmware update?
Any advice on how to proceed? My next plan was to start digging around the BLE API to find some simple functions to verify connectivity to the BLE module and to check the firmware version.
Best regards,
Brett

apalmieriGH, 2017-09-12:
Dear Brett,
first of all thanks for using STM products.
Regarding your setup, please consider that the ST-NUCLEO-F446RE requires a small HW patch as warned in the [X-NUCLEO-IDB0XA1 component](https://developer.mbed.org/teams/ST/code/X_NUCLEO_IDB0XA1) page.
Alternatively, you could even use your NUCLEO board as is provided that you select the ST-NUCLEO-F401RE before building the program from your mbed workspace.
Hope that helps.

Kind regards
Andrea

waymond91, 2017-09-12:
Thank you for the help! 
I am trying this right now. For some reason I assumed this was only required for a literal connection to an arduino.

waymond91, 2017-09-12:
That fixed it!

DavidF13, 2020-03-18:
Hello Brett,

I found this thread trying to fix the same problem you had with the X-Nucleo-IDB05A1. Did you manage to use the shield after the BlueNRG-MS update?